tscircuit/jscad-electronics #57: [Feat]: Add Issue Templates

abhishekHegde2000, 2024-10-04:
**Description:**
Create issue templates to standardize the format and ensure all necessary information is provided for different types of issues. These templates will be added to the `.github/ISSUE_TEMPLATE/` directory.

**Tasks:**
1. **Create Directory:**
   - If it doesn't already exist, create the `.github/ISSUE_TEMPLATE/` directory.

2. **Create Bug Report Template:**
   - Create a `bug_report.yml` file in the `.github/ISSUE_TEMPLATE/` directory with the necessary template content for bug reports.

3. **Create Feature Request Template:**
   - Create a `feature_request.yml` file in the `.github/ISSUE_TEMPLATE/` directory with the necessary template content for feature requests.

**Files to Modify:**
- Create `.github/ISSUE_TEMPLATE/bug_report.yml`
- Create `.github/ISSUE_TEMPLATE/feature_request.yml`

**Benefits:**
- Standardizes the format for submitting bug reports and feature requests, improving clarity and completeness.
- Facilitates easier and more consistent issue tracking and management within the project.

abhishekHegde2000, 2024-10-04:
@seveibar , Hey Can you please assign this task to me?

seveibar, 2024-10-04:
We allow issue template PRs except from maintainers who are familiar with how to create very specific ones for the project

closed 2024-10-04 by seveibar